bancika/diy-layout-creator #588: Cant load diylc on 2009 mackbook pro osx 10.11.6

CaptainChunkulus, 2020-09-29:
Every time I try to open latest version DIY LC, it starts to open then closes and says this application requires a Java runtime environment 1.6.0_10. I downloaded the latest Java but it continues to say this.

bancika, 2020-10-05:
Yes, for some reason old macs refuse to work with Java 8...I suggest sticking with the latest 3.xx version before the Java8 upgrade.

closed 2020-10-05 by bancika

htor, 2020-10-10:
FIY, using java 8 should be no problem on old macs. i use Java 8 and DIYLC 4.9 on my 13" mid-2014 macbook pro.

```
$ /usr/libexec/java_home -V
Matching Java Virtual Machines (1):
    1.8.0_261, x86_64:	"Java SE 8"	/Library/Java/JavaVirtualMachines/jdk1.8.0_261.jdk/Contents/Home

/Library/Java/JavaVirtualMachines/jdk1.8.0_261.jdk/Contents/Home
```

the problems lies in DIYLC being picky about its Java runtime version which it expects to be installed by the user correctly. now, there is nothing wrong with it being picky about the JRE version it wants, but expecting users to provide that exact version is going to result in a lot of these issues on all platforms.

@bancika have you considered https://github.com/bancika/diy-layout-creator/issues/576#issuecomment-693528725?

bancika, 2020-10-10:
> @bancika <https://github.com/bancika> have you considered #576 (comment)

I tried but it didn't work. It's pretty badly documented. It doesn't work
for me and I couldn't find information how to fix it, so I gave up.