bancika/diy-layout-creator #589: Single coil coil tap

wulftone, 2020-10-03:
Nice software! I would love it if there was a single coil with coil-tap component.  Some single coils have three wires coming out (e.g. [Seymour Duncan](https://www.seymourduncan.com/images/wiring-diagrams/3STP_5W_3mtTP_1V_2T.jpg)).  I don't see a way to do this in the software as of right now.

bancika, 2020-10-04:
You can do it. All pickups can have 4 terminals...the coild tap output is
just the two of those terminals joined together.

wulftone, 2020-10-04:
I didn't know you could keep adding "flexible leads".  I did that, deleted one of the two it added and that worked fine.  Is that what you meant?  It looks like I can keep doing that over and over, without a limit?

closed 2020-10-04 by wulftone

bancika, 2020-10-05:
not sure I understand

wulftone, 2020-10-05:
I couldn't find the option to have 4 terminals and still use a single-coil pickup.  A coil tap isn't the same as a humbucker coil split, electronically, so I didn't want to represent it with a humbucker (even a single-coil-sized humbucker).  I discovered that the option to "Add flexible leads" can be repeated and it'll keep adding leads, so that works okay.

bancika, 2020-10-05:
try this

[image: image.png]

wulftone, 2020-10-05:
Looks like your image failed to upload...

bancika, 2020-10-05:
change polarity of pickup to humbucking.