bancika/diy-layout-creator #590: "java.lang.NoSuchFieldError: Splash" Crash on startup, Mac

mstratman, 2020-10-14:
Mac OS X 10.12.6. On startup it crashes with the following:

```
Exception in thread "main" java.lang.NoSuchFieldError: Splash
	at org.diylc.DIYLCSplash.getSplash(Unknown Source)
	at org.diylc.DIYLCSplash.<init>(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
```

This is recreated by manually running `./DIYLC.app/Contents/MacOS/JavaAppLauncher`

The icon appears in the dock momentarily, then disappears. The same dock behavior occurs when right clicking the icon and choosing "Open" (although when opening through Finder I cannot see any additional info about why it quit).
 
No ~/.diylc or logs.

bancika, 2020-10-14:
What is your java version?

mstratman, 2020-10-14:
1.8.0_40

```
/usr/libexec/java_home -V
Matching Java Virtual Machines (1):
    1.8.0_40, x86_64:	"Java SE 8"	/Library/Java/JavaVirtualMachines/jdk1.8.0_40.jdk/Contents/Home

/Library/Java/JavaVirtualMachines/jdk1.8.0_40.jdk/Contents/Home
```

bancika, 2020-10-15:
thanks, that should be ok.
Does the previous version work for you?

mstratman, 2020-10-15:
3.57.0 works for me.
4.6.0 and 4.9.0 have not worked.

bancika, 2020-10-15:
Oh I see. 3.x version uses Java 1.6 and 4.x are upgraded to java 1.8. On
certain Mac machines with Sierra this causes issues that I don't know how
to solve yet. On Mojave it works fine. Go figure.

bancika, 2020-11-19:
can you try the latest 4.10.0 version please?

mstratman, 2020-11-19:
Seems at a quick glance to work! I haven't yet used it extensively but nothing seems amiss.

Thanks for looking into this. 

In case it's helpful I logged the output here: https://gist.github.com/mstratman/26c5201964727e607baa1c2fd1eaa2fe

I'm closing but feel free to reopen if any of those exceptions are related or troubling.

closed 2020-11-19 by mstratman

bancika, 2020-11-19:
Great, thanks for the update. There are still some relatively minor issues
with some mac deployments (like the language-related errors), but it's much
better.