bancika/diy-layout-creator #592: Switching between single and multiple monitors causes 100% CPU usage on OS X

0xhaggis, 2020-10-16:
[log-2020-10-15 09.53.44.txt](https://github.com/bancika/diy-layout-creator/files/5392478/log-2020-10-15.09.53.44.txt)

I frequently switch between using my MacBook as a single standalone device and using it with a large monitor. Whenever DIYLC is open and I do the switch (laptop -> monitor or monitor -> laptop), it causes DIYLC to suddenly max out CPU usage and it won't stop until I kill the DIYLC process.

I don't see anything in the logs related to this, but they are attached. The behavior is NOT intermittent - it happens every time, and it's happened with every version of DIYLC I've tried over the last year or so.

Steps to reproduce:

* Connect external monitor to MacBook
* Run DIYLC and open a file to work on it
* Leaving DIYLC running, unplug the external monitor
* Boom, 100% CPU utilization. Laptop grinds to a halt and the only way to fix it is to kill/restart the DIYLC process.

bancika, 2020-10-16:
Try upgrading to 4.9.0 and see if it helps

0xhaggis, 2020-10-16:
Still does it.

* Upgrade to 4.9
* Load DIYLC
* Load file to work on
* Unplug external monitor
* Instantly pegs the CPU and laptop grinds to a halt until DIYLC is killed.

[log-2020-10-16 10.22.24.txt](https://github.com/bancika/diy-layout-creator/files/5393006/log-2020-10-16.10.22.24.txt)

EL34xyz, 2020-10-22:
Is hardware acceleration checked in the config menu?

I had an issue a long time ago with that on Windows 10 dual monitor system
Just a wild guess on my part?

0xhaggis, 2020-11-08:
> Is hardware acceleration checked in the config menu?
> 
> I had an issue a long time ago with that on Windows 10 dual monitor system
> Just a wild guess on my part?

That worked. Turning off hardware acceleration makes it possible to switch monitors in and out with no issues.

EL34xyz, 2020-11-08:
Nice!

closed 2020-12-11 by bancika