bancika/diy-layout-creator #593: Im already instal DIYLC on Mac OS X high Sierra, and does not appear nothing , the only thing apear on de top left corner the menu DIYLC for 4 seconds and disappears.

maiagab, 2020-10-26:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-10-26:
Sierra has some issues with java 8 and DIYLC. You can use the latest 3.xx
version of DIYLC

bancika, 2020-11-19:
can you try the latest 4.10.0 version please?

closed 2020-11-19 by bancika