bancika/diy-layout-creator #594: pin header rows and columns are swapped

htor, 2020-11-15:
currently, when editing the Pin Header component (shown in the picture) the "Rows" field reflects the actual number of columns, not rows, and vice versa with "Columns". i would expect a change of rows would change the y-axis and a change of columns a change in the x-axis.

for instance: here it says 5 columns and 2 rows, but it should be 2 columns and 5 rows.
[image]

bancika, 2020-11-19:
Actually it's kind of intended like this. The way the header will be rendered also depends on the orientation settings, so we can't guarantee that rows will be vertical and columns will be horizontal anyways. The 10pin header you created is referred to as "2 row" header on Mouser and some other sites, so that's why I followed that logic. Maybe the default orientation could've been horizontal, but that wasn't consistent with ICs and switches.

closed 2020-11-19 by bancika

htor, 2020-11-20:
allright, i see now!