bancika/diy-layout-creator #595: What is the newest iOX version usable for DIYLC?

Swampertech, 2020-11-19:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-11-19:
what is iOX?

Swampertech, 2020-11-19:
It’s the latest version of Mac software. I am currently using Catalina, but there is a new update. I don’t want to update if I could loose the use of your great program. Thanks!

bancika, 2020-11-19:
oh you mean OSX...yeah, I think it should work, but no guarantees.

bancika, 2020-11-19:
can you please give v4.10.0 a shot? I just published it. It contains
bundled JRE, so less work around the installation and should be safer.

Swampertech, 2020-11-19:
Has this version (4.10.0) been uploaded on the new Mac software Big Sur and proved to work?

bancika, 2020-11-19:
Hasn't been tested on the latest OS. I recommended that version because it
includes java version needed for it to run, so it's more likely to work
without additional actions.

Swampertech, 2020-11-19:
Hey Bane, I am running DIYLC v3.57, which I managed to get to run on OS Catalina (had the Java problems). Anyhow, I wouldn’t want to loose my DIYLC program. So does the version 4.10.0 will work on Catalina? Thanks again for your help.

bancika, 2020-11-19:
4.10 works on my latest Catalina with no issues. If you are upgrading from
3.xx version I suggest you to make a backup of your config file, in case it
gets damaged during the upgrade.

It's located in a (hidden) directory in /Users/<yourname>/.diylc/config.xml

Just make a copy of that file in case you want to go back to 3.xx version,
so you keep your settings. Nothing else is needed really.

4.10.0 comes with Java bundled inside to avoid any issues with Java.

Cheers

htor, 2020-11-21:
> 4.10.0 comes with Java bundled inside to avoid any issues with Java.

hooray!