bancika/diy-layout-creator #598: Label Editing Causing Freezing/Crashing

StrangeAudio, 2020-11-26:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

StrangeAudio, 2020-11-26:
Found an issue based around creating and editing labels specifically. When I double click on an existing label, or right click and edit, the program freezes and does not unfreeze. I've found this issue on the last three or four versions, and just tried the newest (4.10.0) and the problem is still present. 

Issue does NOT happen when manipulating labels outside of editing them, i.e. I can move labels, delete them, copy/paste, etc. 

I'm running this on Windows 10, version 2004. I've attached the last few log files. 

Please advise. 

[log-2020-11-26 09.09.21.txt](https://github.com/bancika/diy-layout-creator/files/5603644/log-2020-11-26.09.09.21.txt)
[log-2020-11-26 09.08.46.txt](https://github.com/bancika/diy-layout-creator/files/5603645/log-2020-11-26.09.08.46.txt)
[log-2020-11-26 09.42.09.txt](https://github.com/bancika/diy-layout-creator/files/5603646/log-2020-11-26.09.42.09.txt)
[log-2020-11-26 09.11.36.txt](https://github.com/bancika/diy-layout-creator/files/5603647/log-2020-11-26.09.11.36.txt)

bancika, 2020-11-28:
Few questions:

1) does it recover after a few seconds?
2) if it recovers, does the next attempt work fine?

There's an issue with font drop down on some machines that causes it to
slow down the dialog but only the first time.

StrangeAudio, 2020-11-28:
It does not recover, program stays frozen indefinitely. Only way out is to force close/end task from Task Manager. Then reopen program from last saved backup file.

So far this happens every time I try to edit text, without exception.

bancika, 2020-12-18:
How many fonts do you have installed in the system?

StrangeAudio, 2020-12-18:
Looks like 122 total fonts on this machine. I don't believe I have any out-of-the-ordinary fonts, nothing that isn't standard with Microsoft or Adobe programs, for instance.

StrangeAudio, 2020-12-18:
I just downloaded the newest update of DIYLC, went into a file, right clicked on a label, and selected "Edit Selection" and it still freezes indefinitely.  Here is the log file from that. 

[log-2020-12-18 06.48.47.txt](https://github.com/bancika/diy-layout-creator/files/5718580/log-2020-12-18.06.48.47.txt)

bancika, 2020-12-19:
Hm, so font number doesn't seem to be the issue. I have twice as much and no issues at all. It seems that font chooser box that shows preview of all fonts is to blame.

Maybe I can make a small test app that logs every step of creating font chooser. That way you could run it and it would tell us when it stops or slows down dramatically.

Would you be up for that?

Cheers,
Bane

StrangeAudio, 2020-12-19:
I'm cool with that. Anything to figure out the issue.

bancika, 2020-12-19:
Here's a zip file. Please unpack anywhere and run the bat file. It will do pretty much the same thing that is done in DIYLC but step by step. You will see in the console how far it got. Please check if it runs to completion and send me the whole text output from the console.

[font test.zip](https://github.com/bancika/diy-layout-creator/files/5719520/font.test.zip)

StrangeAudio, 2020-12-19:
I just ran it and the console was up for about a second and then closed out.

bancika, 2020-12-19:
hm, then it didn't work at all...hmm. Here's another one that catches more errors, please try again.

Also, it's better to run from command prompt because it won't close on error and you will see the reason

here's how

`Microsoft Windows [Version 10.0.19041.685]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\bancika>cd Desktop

C:\Users\bancika\Desktop>run.bat`

[fonts2.zip](https://github.com/bancika/diy-layout-creator/files/5719536/fonts2.zip)

StrangeAudio, 2020-12-19:
Still seeing an error. Here's the info from command prompt:

[image]

bancika, 2020-12-19:
you extracted zip on Desktop right? you can see a file named fonts.jar?

StrangeAudio, 2020-12-19:
My bad, I unzipped in another folder and didn't drag it all together. Running it now.

StrangeAudio, 2020-12-19:
Here's the result from the scan:

[Fonts Result.txt](https://github.com/bancika/diy-layout-creator/files/5719558/Fonts.Result.txt)

bancika, 2020-12-19:
That's great, thanks! It pinpoints the issue to font preview rendering. Wow, 200 seconds to render 390 fonts, that's crazy. It takes 1-2 sec on my machine for 260.

Can we please try one more experiment. This time I'm timing each font separately to see if there's any problematic one that takes much more than others. That will tell us if font rendering is slow for some reason in general or there's a font that causes issues.

Thanks again

[fonts3.zip](https://github.com/bancika/diy-layout-creator/files/5719570/fonts3.zip)

StrangeAudio, 2020-12-19:
Here's the results from the second test. You see some serious bogging down on some fonts for some reason for sure. 

[Fonts Result.txt](https://github.com/bancika/diy-layout-creator/files/5719591/Fonts.Result.txt)

bancika, 2020-12-19:
Thanks, that's great results. Some of the fonts are crazy slow while others are in the same ballpark as on my machine.
This one is the worst, 20 seconds.  Kozuka Gothic are almost as bad. I'll download some of them on my machine to see if they behave the same way.

Font: Adobe Myungjo Std M
Font rendered in 20030ms

StrangeAudio, 2020-12-19:
Would it be as simple as deleting some of the worst offenders? I can't imagine needing a Japanese font anytime soon.

bancika, 2020-12-19:
Sure

StrangeAudio, 2020-12-19:
I deleted a handful of fonts that were the slowest to load, and I'm down to 11 seconds for the whole scan. Way better. 

More importantly, I opened up DIYLC, double clicked on a label, and after 5-6 seconds it loads and I can edit, no freezing. Went through and edited a bunch of other things too, and it seems to have solved the problem. 

Guess the bigger question is why those particular fonts are taking so long to load?

bancika, 2020-12-19:
They seem to be much bigger. While regular fonts are 50kb or so, these are
3-4mb each. Probably has to do with that.

bancika, 2020-12-19:
I have an idea. I will make DIYLC analyze all fonts in background the same way we just did and identify
 those that are slow and not render them in the preview. Here's a DIYLC build that attempts to do that. Can you please extract it somewhere isolated from your main installation and run it to see if it's any better?

[diylc-4.13.0.zip](https://github.com/bancika/diy-layout-creator/files/5719700/diylc-4.13.0.zip)

Thanks

StrangeAudio, 2020-12-19:
That seemed to do the trick. No hesitation in editing or creating labels. I tried it both with a brand new DIYLC file, and an existing one, and things are smooth.

bancika, 2020-12-19:
nice. I'll polish it up and include in 4.13

thanks for your help, this one has been chasing me for years.

closed 2020-12-19 by bancika