bancika/diy-layout-creator #599: I can't import xml files from Hoffman on my Mac

brewdude66, 2020-11-30:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

brewdude66, 2020-11-30:
What is "no args constructor"?
It seems that what ever it is, not having it is keeping the xml variants from loading 
Am I doing something wrong?

brewdude66, 2020-11-30:
I'm sorry, but I can't seem to find the diilc/logs on my MacBook--the search function is broken.

EL34xyz, 2020-12-06:
I just tested importing this file from this web page
HoffmanBuildingBlocks.xml

https://el34world.com/charts/DiyLayoutCreator/files/Template_Files/Template_Files.htm

The building blocks all import fine on my Win 10 machine
I don't know anything about MAC's
Maybe someone else can test importing the building blocks

Just to make sure you are importing the correct file
You are importing the .xml file, not the .diy file?

DIYLC logs files on Windows 10 are in this folder
C:\Users\hoffm\.diylc
User name hoffm would be different on other machines

I attached a screen shot of my logs folder

[image]

[image]

brewdude66, 2020-12-07:
Thanks for the help.
I did try to load the building blocks xlm, just like instructed.  
I believe it must be a problem with the Mac and it’s OS. 

I would be grateful if the issue was resolved, but I am learning to build my own variants. 

I appreciate the help
Thanks,
Lance McLaughlin

bancika, 2020-12-13:
Please upload one of the log files created at the time you encountered the issue. You can grab logs by going to DIYLC -> Help -> Access User Files, then go to logs folder.

bancika, 2020-12-31:
Actually, I was able to reproduce the issue on my machine with the latest DIYLC. Seems like I changed something almost two years that made some compatibility issues with older building blocks files. I updated Hoffman's file to work with the latest version of the code, but will need to make a small code patch too in 4.15 version that's pending the release.

@EL34xyz it's strange that it works for you

[HoffmanBuildingBlocks.xml.txt](https://github.com/bancika/diy-layout-creator/files/5757181/HoffmanBuildingBlocks.xml.txt)

brewdude66, 2020-12-31:
Should I wait for the 4.15 update?
Thank you for all the effort.  
Lance

bancika, 2020-12-31:
Yes, please do.

brewdude66, 2021-01-04:
I downloaded version 4.15, but it won’t complete loading on my machine… It just stalls on the opening graphic.
I tried putting the previous version (4.13) in the trash bin, and loading the version again, but still stalls at the opening.

bancika, 2021-01-04:
Please try the latest version

bancika, 2021-01-09:
Assuming all good

closed 2021-01-09 by bancika

brewdude66, 2021-01-09:
I have not tried the 4.16.0 update yet, but 4.15.1 seems to have worked. 

Regarding the updates, is there a way to update with out replacing?

I have found it somewhat cumbersome to have the multiple versions and then have to figure out which to trash. 

Should I delete the previous version before downloading the new one?

Thank you for all the help,
Lance

EL34xyz, 2021-01-09:
If you download the non installable versions, you can just run them as stand alone.