bancika/diy-layout-creator #602: Menubar destroy in german language

HaraldRau, 2020-12-09:
Hello,
In the German language there is a problem with the display in the menu bar. The desire for coffee destroys the menu bar.

[log-2020-12-09 12.57.09.txt](https://github.com/bancika/diy-layout-creator/files/5665696/log-2020-12-09.12.57.09.txt)

[image]

Thank You for your perfect Work!

bancika, 2020-12-09:
Hi Harald, thanks for the heads-up and thanks for the solution! I have included it in the next release.
And thank you for the coffee, it's highly appreciated!

closed 2020-12-09 by bancika

HaraldRau, 2020-12-09:
Hello, no problem. Enjoy it!
It's a great program. I used it to design the circuit board for my
simple tube amplifier.