bancika/diy-layout-creator #605: DIYLC 4.12 won't start on Linux Mint 20

Patrick-Knopf, 2020-12-15:
I've installed DIYLC 4.12 under Linux Mint 20 Cinnamon 4.6.7, kernel 5.4.0-58-generic and openjdk 1.8.0_275.
Trying to start DIYLC with ./run.sh or sh run.sh shows the splash screen but nothing else. On the command line I can see that it stops running after that ->
```
14:15:23,204  INFO [main] Presenter - installPlugin(CanvasPlugin)
14:15:23,226  INFO [main] CanvasPlugin - Magnification gesture listener initialized.
14:15:23,232  INFO [main] MainFrame - injectGUIComponent(org.diylc.swingframework.ruler.RulerScrollPane, 0, false)
14:15:23,248  INFO [main] Presenter - installPlugin(ComponentTree)
14:15:23,332  INFO [main] MainFrame - injectGUIComponent(org.diylc.swing.plugins.tree.TreePanel, 2, true)
14:15:23,396  INFO [SwingWorker-pool-1-thread-3] UpdateChecker - 0 updates found
14:15:24,190 DEBUG [SwingWorker-pool-1-thread-2] PhpFlatProxy - Connecting to: http://www.diy-fever.com/diylc/api/v1/getAnnouncements.php
14:15:24,342  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - Trying to download file: http://diy-fever.com/diylc/api/v1/update.xml
14:15:24,531  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 0 updates found
```
Any hints what to do?

Kind Regards,
Patrick

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-12-15:
Does any older version work? Maybe that could help pinpoint the issue.

Patrick-Knopf, 2020-12-15:
Any DIYLC 4.x version won't work, that all stuck at the UpdateChecker.
The DIYLC 3.57 works fine...

bancika, 2020-12-15:
seems to be java related. DIYLC 4 is upgraded from java 6 to 8.

can you try to change run.sh to include this argument to java when launching DIYLC

-XX:ErrorFile=hs_err_pid_%p.log

if there's any java errors they should show up in that log.

bancika, 2020-12-15:
Tried it on Mint VM and it works no issues. Java version is the default one
that ships with Mint 20: 11.0.7+10-post-Ubuntu-3ubuntu1 by Ubuntu

[image: image.png]

Patrick-Knopf, 2020-12-15:
Yes, confirmed. I've changed my Java to version 11 and it works here as well. I had used version 8 for JOSM, but it seems that they also support version 11 now.

Thanks a lot for your support
- Patrick

closed 2020-12-15 by bancika