bancika/diy-layout-creator #608: MacOS version stuck at splash screen

yorkshirelandscape, 2020-12-23:
The current build won't run on MacOS High Sierra. It just hangs at the splash screen. I can quit it normally, but can't do anything in it. Here's the log:

08:54:24,198  INFO [main] ConfigurationManager - Initializing configuration for: diylc
08:54:25,119  INFO [main] ConfigurationManager - Saving configuration
08:54:25,413  INFO [main] DIYLCStarter - Loading languages...
08:54:25,415  INFO [main] LangUtil - Loading translation for English
08:54:25,445 DEBUG [main] LangUtil - Loading language file: /Applications/DIYLC.app/Contents/Resources/lang/English.txt
08:54:25,504 DEBUG [main] DIYLCStarter - Java version: 1.8.0_275-b01 by AdoptOpenJDK
08:54:25,504 DEBUG [main] DIYLCStarter - Java home: /Applications/DIYLC.app/Contents/PlugIns/jdk8u275-b01-jre/Contents/Home
08:54:25,504 DEBUG [main] DIYLCStarter - OS: Mac OS X 10.13.6
08:54:25,504  INFO [main] DIYLCStarter - Starting DIYLC with working directory /
08:54:25,504  INFO [main] DIYLCStarter - Configuration dump start.
08:54:25,506  INFO [main] DIYLCStarter - highlightContinuityArea = false
08:54:25,506  INFO [main] DIYLCStarter - defaultTemplatesImported = true
08:54:25,506  INFO [main] DIYLCStarter - defaultBlocksImported = true
08:54:25,506  INFO [main] DIYLCStarter - Configuration dump end.
08:54:25,511  INFO [main] DIYLCStarter - Creating the main frame...
08:54:25,512  INFO [Thread-1] FontOptimizer - Starting font optimizer
08:54:26,494  INFO [main] Presenter - Current DIYLC version: 4.13.0
08:54:26,519  INFO [main] Presenter - Checking if variants need to be updated
08:54:26,520  INFO [main] Presenter - Loading component types.
08:54:26,521 DEBUG [main] Utils - Scanning jar: /Applications/DIYLC.app/Contents/Java/main.jar
08:54:27,261 DEBUG [main] Utils - Scanning jar: /Applications/DIYLC.app/Contents/Java/diylc-core.jar
08:54:27,299  INFO [main] Presenter - Loading additional library JARs
08:54:28,994 DEBUG [main] JarScanner - Found JAR file: WCFirefoxExtn.jar
08:54:29,103 DEBUG [Thread-1] FontOptimizer - Fonts loaded in 0ms
08:54:30,264  INFO [Thread-1] FontOptimizer - Font optimization completed in 1158ms. Found 14 slow fonts
08:54:30,265 DEBUG [Thread-1] FontOptimizer - Slow fonts: Nadeem,Trebuchet MS,Kailasa,Geeza Pro,Al Bayan,Beirut,Sathu,Devanagari MT,Gurmukhi MT,Raanana,Chalkduster,STIXIntegralsUp,.SF NS Text,Arial Hebrew Scholar
08:54:30,265  INFO [Thread-1] ConfigurationManager - Saving configuration

bancika, 2020-12-23:
Does the previous version work ok?

yorkshirelandscape, 2020-12-23:
Yes, it does.

bancika, 2020-12-23:
Great, thanks! I'll investigate. In the meantime stay with the old one.

hcornel, 2020-12-27:
Helpful report. And in case it helps too, I get exactly the same thing with 4.13 under Big Sur OS 11.

bancika, 2020-12-27:
Can you try to wait 10-15mins if anything happens? On one of my VMs it
hangs for ~10mins and then starts working again. Just want to confirm it's
the same issue.

hcornel, 2020-12-27:
Yes, that works. I didn't sit and time it, but when I came back in ~ 15 minutes it had opened all the way. Hope this helps.

bancika, 2020-12-27:
Cool, thanks! Will look for the root cause.

bancika, 2020-12-27:
Can you please upload the latest log file created at the time you let it
open all the way?

bancika, 2020-12-27:
Also, can you please try this version? I tried some changes that could help

https://easyupload.io/yfpuec

hcornel, 2020-12-27:
Re the log, I'd be happy to send ,but I don't know how to find it on my Mac.

I will try the new version. Thanks.

hcornel, 2020-12-27:
And, yes, the new version opens right up. Thank you!!!

closed 2020-12-28 by bancika