bancika/diy-layout-creator #609: Hangs when application left open

pmfawcett, 2020-12-24:
4.13.0 (and I think 4.12.0 was doing the same thing) will hang if I leave the application open overnight while working on a project.   When I return Javaw.exe will be burning about 12-13% CPU, and the app is frozen.  This is on Win 7.  Note that I am not putting the machine to sleep or hibernate.

[log-2020-12-21 12.41.00.txt](https://github.com/bancika/diy-layout-creator/files/5740385/log-2020-12-21.12.41.00.txt)

bancika, 2020-12-29:
Hm, I tried to replicate with one of my projects and it was working fine the next day. Maybe it has something to do with the project? Can I try yours?

pmfawcett, 2020-12-29:
I have uploaded project "B15N lite" to the cloud so that you may test.  More recently I've noticed that it doesn't always hang.  It is possible (this is speculation) that the issue has to do with memory allocation / usage? Over time an unreasonable-seeming amount of memory can get tied up by DIYLC at the best of times,, even if GC is forced.  Note also that I very often I will also have running the JetBrains PyCharm IDE executable, which is likewise memory hungry, and I believe is also a java-based app.  Could there be an issue here (even if overall system memory use should be acceptable)?

bancika, 2020-12-29:
Thanks, Paul. It's possible that it's memory related. Earlier this year I added some caching that improved speed dramatically, but now it takes slightly more memory. I tested it quite a lot and under my regular usage, most if not all memory gets GC'd. Now, if there was a scenario that leads to a memory leak, it would cause a pile up and it would slow down eventually.
I wish I could see it running through the profiler when it happens, it would help pinpoint the issue.

bancika, 2020-12-29:
In your case, it's not the cache to blame, according to the log

14:09:02,148 DEBUG [AWT-EventQueue-0] DrawingCache - Render cache contains 1199 elements, approx size is 185 MB

pmfawcett, 2021-01-02:
I've had an occurrence again this morning. Took a 2-3 days to hang this time. Total memory use was only about 700,000K, but CPU varying between 13-18%.  Note that when the application is unresponsive this way, if the window is dragged or moved it does not re-render, it just replaces with black. This time I did NOT have the PyCharm executable running, but there were some chrome windows that looked like they had leaked and consumed a large amount of memory. However killing these did not restore DIYLC.

pmfawcett, 2021-01-03:
Another hang today, but now confirming that it will crash just the same with a new "untitled" file with no edits made.. Again, completely unresponsive with 600,000+K memory, and CPU burning ~16%. This time there were only a couple of other windows open in Chrome, with absolutely nothing remarkable about memory use or otherwise.  AVG antivirus I think the only other thing running.   Log file seems to be uninformative.

[log-2021-01-02 10.53.31.txt](https://github.com/bancika/diy-layout-creator/files/5761824/log-2021-01-02.10.53.31.txt)

Java(TM) Platform SE binary 8.0.2110.12

bancika, 2021-01-03:
Cool. Can you try toggling the hardware acceleration setting and see if it
works?

pmfawcett, 2021-01-03:
OK, I normally have hardware acceleration "ON" by default, so I turned it "OFF" and will let it sit and think, see what happens this time.

bancika, 2022-11-28:
Is this still an issue? In the meantime I disabled hardware acceleration because others had similar issues with it enabled. Closing, please reopen if needed

closed 2022-11-28 by bancika