bancika/diy-layout-creator #613: Windows 10 cannot open diy file in 4.14.0v

shugoPL, 2020-12-29:
Hi,
I can`t opne diy files on windows 10 platform.
18:24:09,632  INFO [SwingWorker-pool-1-thread-4] ProjectFileManager - loadProjectFromFile(**[path_to_file]**\ Little Gem-Amp.diy)
18:24:09,638 ERROR [SwingWorker-pool-1-thread-4] Presenter - Could not load file
java.lang.IllegalArgumentException: Unknown file format version: 
	at org.diylc.serialization.ProjectFileManager.deserializeProjectFromFile(Unknown Source)
	at org.diylc.presenter.Presenter.loadProjectFromFile(Unknown Source)
	at org.diylc.swing.ActionFactory$OpenAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.ActionFactory$OpenAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.gui.MainFrame$6.doInBackground(Unknown Source)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
18:24:11,575  INFO [AWT-EventQueue-0] CanvasPlugin - Pressed: false

bancika, 2020-12-30:
please upload the file so I can examine it (you may need to change the extension to txt)

shugoPL, 2021-01-03:
Sory for so long answer. 
[[Amplifier] Little Gem-Amp.txt](https://github.com/bancika/diy-layout-creator/files/5761342/Amplifier.Little.Gem-Amp.txt)

bancika, 2021-01-03:
hey, thanks. That's V1 file over a decade old. There was a bug in the code
that prevented it from being loaded. I'll include the fix in 4.15 version
that will go out probably later today.

Cheers

closed 2021-01-03 by bancika