bancika/diy-layout-creator #615: Flatpak build

LeandroStanger, 2021-01-03:
Please, consider supporting [Flatpak](http://docs.flatpak.org/en/latest/introduction.html#reasons-to-use-flatpak) and publishing DIY Layout Creator on [Flathub](https://github.com/flathub/flathub/wiki/App-Submission).

bancika, 2021-01-03:
Sure, sounds good, but I know nothing about Flatpack.
DIYLC is a Java Swing desktop app that builds using oldschool ANT files.
Do you have some boilerplate examples for working with java apps? Can Java
JRE be bundled with the package?
If you take a look at this file, it's the whole app without the Java JRE
and it runs on all major OSs. Any tips on packing that with flatpack?

https://github.com/bancika/diy-layout-creator/releases/download/v4.14.0/diylc-4.14.0.zip

ChrisOfBristol, 2021-01-03:
Sorry Bane, on Flathub it looked like this guy was offering help. I don't know why he has posted this as my post on Flathub had the mandatory field "Upstream has been contacted: Yes"!
Chris

closed 2021-01-09 by bancika