bancika/diy-layout-creator #618: Corrupted File

mmihalj, 2021-01-04:
In version, 4.15.1 on Win10.
Pressed Ctrl+D and DIY Layout Creator froze, garbage collector went red, and file I was working on got corrupted.

bancika, 2021-01-04:
can you please attach the log file from the time it happened? You can find
it in c:\users\<you>\diylc
In the same folder you can find backups so you can restore your file to the
last saved state.
Cheers

mmihalj, 2021-01-07:
Thank you. That helped. 

I am attaching logs, and latest backup as well as the file that appears corrupted. Maybe that will indicate reasons for this issue.

 

Thank you Bane.

 

 

Hvala na podrsci I topli pozdrav iz TX.

 

Miroslav

bancika, 2021-01-07:
Seems that none of the attachments made it through, it doesn't work through email replies.

Pozdrav iz Nisa :)

mmihalj, 2021-01-08:
E jebes poso da ti ja kazem. Ne znam kako da ti posaljem attachements onda. 

Nij enista hitno alia ko imas neki drugi email mogu da saljem tamo…

 

Gledo sam da zakazim direct na GITHUB nema ni tamo

 

Daj neki drugi email ili dropbox ili sta god ti je po volji.

Ovaj nece izgleda da prima attachements: Bane Stojkovic notifications@github.com <mailto:notifications@github.com> 

 

Ukratko – pmogao si mi nasao sam backup ali kontam da mozda moj primer moze da pomogne stabilnosti programa. 

 

 

Ziv bio, gitare se nasvirao!

 

 

M

bancika, 2021-01-08:
Moze na bancika @ gmail.com
Hvala

bancika, 2021-01-09:
Okay, I think I see the problem. There's a bug with Wrap Label component that makes it hang the app. You can go to your corrupt file and just remove the last wrapLabel component at the end of the file and it will work.
I'll seek the root cause.

bancika, 2021-01-09:
On second look, I think I see the problem. You made a wrap label with 0 length, so it tried to wrap text to 0 length and got in an infinite loop. If you change these two lines at the end of the file it will work again

    <diylc.misc.WrapLabel>
      <name>L1</name>
      <controlPoints>
        <point x="0.9" y="5.6"/>
        **<point x="2.0" y="5.7"/>**
      </controlPoints>
      <firstPoint x="0.9" y="5.6"/>
      **<secondPoint x="2.0" y="5.7"/>**

bancika, 2021-01-09:
fixed the bug for 4.16

closed 2021-01-09 by bancika