bancika/diy-layout-creator #619: PLEASE, please give us an option to darken the grid lines.  I love the app but it gives me eyestrain.  Thank you!

MarsManiac, 2021-01-08:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2021-01-08:
you can make your own theme with customized colors. Go to diylc -> themes folder, copy one of them and change colors and name of the theme. It should appear in Config -> theme menu in DIYLC after restart.

closed 2021-01-08 by bancika

MarsManiac, 2021-01-08:
Got it.  Thanks.  Great app!

MarsManiac, 2021-01-22:
Hi, 

I was first able to darken the grid color but I downloaded the newest version today (1/22/21) and lost that edit when the new config.xml copied down.  I’m on macOS Big Sur.  

I can edit the XML file to create another theme entry but when I restart DIYLC I don’t have the new theme as an option in the Config—>Themes menu.  Am I editing the wrong config.xml?  (Very familiar with editing XMLs overall)

Can you tell me the exact Mac folder where I should be able to find the config.xml file?

Also, the grid is not showing up on my workspace.  I see it in the larger ‘expanded’ workspace but my project work area is solid white.  Any suggestions to fix that?

Hoping these are known questions with easy answers.

Thanks,

Richard

Closed #619 <https://github.com/bancika/diy-layout-creator/issues/619>.