bancika/diy-layout-creator #622: Alpha setting on Color attribute not being saved

rsmilward, 2021-01-16:
Tried to save a tripad board variant with the "Color" set to white with RGB Alpha set to 0, but it always comes back as Alpha = 255. Seems to apply to CMYK Alpha as well.

bancika, 2021-01-16:
Yes, this is intentional. Color is only for color setting, and there's a separate Alpha setting that works

[image]

rsmilward, 2021-01-16:
> image

But that fades the entire selection, including the traces I want to see!
I can save a variant with a different color Border (not that I need that 
right now).
The Alpha of the Color is also a changeable attribute, so why can't 
/*it*/ be saved?
I can save a Mini Toggle Switch variant with a lighter Body Color, even 
a Transparent one,
and it stays that color when I add a new copy of it (from the menu, not 
making a Copy in place),
so I should be able to save a Tripad board with a transparent color too.
This inconsistency is very frustrating for me...
**Richard

(Also, WTH do I have to *double*-click a part category to open it?
This is so close to being an ideal and easy to use design tool.
Used it several times in my last job with great success --
but didn't need this transparency feature for those projects.
I've just made a $25 donation to indicate my appreciation of your work 
so far.)

rsmilward, 2021-01-16:
Damn -- I donated to DIY-Fever.
I've donated to you too...
**Richrd

On 1/16/2021 4:55 PM, Richard Milward wrote:
> I've just made a $25 donation to indicate my appreciation of your work 
> so far.)

bancika, 2021-01-16:
Ok, I see what you mean now..you want to make only one "area" of the
component transparent but keep the outline and everything else intact.
I see that being useful...I'll try to make that change for the next version.

What do you mean re: click on the category, I didn't get that part?

As for the donation - thank you, I appreciate it. I saw that both of them
made it through to my account, so I can ask paypal to refund one?

All the best

rsmilward, 2021-01-16:
Thanks, Bane. Yes, the tripad and Vero boards have a background which 
hides anything they overlap. I'm trying to duplicate an odd prototype 
board which has quad-pad strips and busses between them -- this one:
https://www.ebay.com/itm/163856537806 
<https://www.ebay.com/itm/163856537806>
I intend to use a couple of them for a retro computer front panel layout 
(blinkenlights & switches!)

In the left-side menu, I have to double-click to open the Boards 
category, for example (or any of those groups). It seems like that could 
be a single click.

I reacted too soon with my donations! I didn't realize that you were DIY 
Fever -and- DIYLC. I'm just in a hurry today and need to slow down... 
Yes, if you could refund one of those donations, I'd appreciate it.

Thanks again, sorry for the rants, the isolation is getting to me.
Here's hoping for the best! And a delayed Merry Christmas and Happy New 
Year too.

**Richard

bancika, 2021-01-16:
Refunded the donation.

I see what you mean re: categories. You must be coming from Mac ecosystem? On windows double click is standard. I'll check if that can be changed in Java.

bancika, 2021-01-17:
Ok, I think I addressed both issues you brought up. It will probably go out
tonight in the 4.16 version.
Cheers

closed 2021-01-17 by bancika

rsmilward, 2021-01-21:
Bane,
I thought you'd like to see the board layout I've been working on - it's 
attached. I know, it's too big for a guitar pedal! :-)

If you select the entire thing and Group it, then you can change the 
Color to white with 100% Transparency, and then you'll see what I've 
been trying to create. I can export this as a PNG and print it with no 
problem.

It may just be this 4.15.1 version I'm still using, but if you just 
select it all (without Grouping) and then try to change the Color, the 
program hangs.

There's a minor color issue I'll report shortly.

Thanks.

**Richard