bancika/diy-layout-creator #624: Can't open files, Mac OS Catalina

Mentolo, 2021-01-21:
Hi!

After upgrading from 3.40 to the latest version, many files won't open.
But some do :) Can you enlighten my path?
I attached a non opening file and the corresponding log.

Cheers and thanks so much!
Phil
[Hermelin 8 Layout 3.1 Tremolo Kopie.diy.zip](https://github.com/bancika/diy-layout-creator/files/5851755/Hermelin.8.Layout.3.1.Tremolo.Kopie.diy.zip)
[log-2021-01-21 08.42.20 Kopie.txt.zip](https://github.com/bancika/diy-layout-creator/files/5851758/log-2021-01-21.08.42.20.Kopie.txt.zip)

bancika, 2021-01-21:
Hi Phil,

You seem to be one of very few people who used the old can capacitor
component that wasn't good and got dropped. One way go get it to work is to
manually edit files in text editor and delete all instances
of org.diylc.components.passive.ElectrolyticCanCapacitor

Cheers

jorgefortes, 2021-01-22:
Hi Bancika

I've already tinkered on this with Phil before. I've now removed all the can capacitors from the XML structure and, lo and behold, the layout is loading. Unfortunately, there is one error message to overcome. I’ve attached a screenshot and the corresponding log.

Thanks for your advice, Bancika.

Cheers
Jorge

[image]
[log-2021-01-22 09.30.33.txt.zip](https://github.com/bancika/diy-layout-creator/files/5854467/log-2021-01-22.09.30.33.txt.zip)

bancika, 2021-01-24:
> [image:
> Error_und_DIY_Layout_Creator_v4_16_0_-__Users_jorgefortes_diylc_Fender_Twin_Reverb_1_1_diy]

do you open files from the finder or through the apps File -> Open menu?
What's the exact path to the file? Seems like the path somethow got
truncated and the only thing left is the -psnXXX crap from the message box.
Maybe it has some special characters in the name?

bancika, 2022-11-28:
is this still an issue?

closed 2022-12-14 by bancika