bancika/diy-layout-creator #625: Color to be edited is not marked in the palette

rsmilward, 2021-01-21:
When I select an item, then go to Edit Selection and pick e.g. the Border color, I'd like to see the current color marked with an X in the Swatch palette.

bancika, 2021-02-04:
I don't think this is possible in Java's color chooser dialog. It shows current color in the bottom part of the dialog, but X's are shown only when clicked.

rsmilward, 2021-02-04:
OK, thank you for checking. I found that the current color is in the RGB 
palette, which is helpful. That might be worth mentioning in the help or 
documentation.
**Richard

bancika, 2021-02-09:
Glad it works for you. As for the documentation, feel free to add anything you think could be useful, the docs are also open source.
Cheers

closed 2021-02-09 by bancika