bancika/diy-layout-creator #628: Wishlist - connectiviy component: 1-core and 2-core screened cable

aabill2, 2021-02-03:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2021-02-21:
There's an easy workaround for this - just combine regular wires of different gauges

[image]

aabill2, 2021-02-22:
Than you, much appreciated

Bill

closed 2021-07-12 by bancika