bancika/diy-layout-creator #629: Have to restart when using "undo" too often

tubeampprofessor, 2021-02-06:
Hey there!

When I use "undo" too often the drawing will freeze. I am no longer able to move, change or add components. Luckily, I can still save my drawing. Still, I will have to close and restart the programm to continue working on it.

Thx!
Georg

[log-2021-02-06 12.01.34.txt](https://github.com/bancika/diy-layout-creator/files/5935965/log-2021-02-06.12.01.34.txt)
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2021-02-07:
Found a memory leak that doubled the memory footprint each time you hit Undo. Fixed for 4.17.0

closed 2021-02-07 by bancika

tubeampprofessor, 2021-02-15:
That was fast :) I just downloaded the latest build. Thank you so much!