bancika/diy-layout-creator #630: It displays <br> instead of a line break in the Auto-Wrap Label

Impartcho, 2021-02-07:
Hi, simple bug here, everything in the title.
Thanks bancika for this awesome soft btw.
(Using v4.16.0)

bancika, 2021-02-07:
Works fine for me. Will need more info - what are the circumstances - text,
context, OS, etc?

Impartcho, 2021-02-07:
OS is Windows 10 Pro x64 and it happens everytime I try to put a line break in a text in the field value of "Misc/Auto-Wrap Label" like in the pic related : 
[https://i.imgur.com/M9nDrQF.jpeg](https://i.imgur.com/M9nDrQF.jpeg)
I thought it was an easy fix, it can be a language issue (Is the text in value incorrectly recognized as HTML? Is < br > used in other languages than HTML?)
Maybe it's more complicated than that...

Absolutely not related, but I found another thing that could make the software better : it's to center the zoom/dezoom (with alt+wheel) towards the middle of the screen/board (and maybe somewhat related to the mouse position) when no object is selected. Maybe just to center it towards the dotted area when nothing is selected would be perfect. It's pretty random now, I always finish with my components somewhere far away from my screen. Anyway it's not a big deal, just an idea to improve UI.

bancika, 2021-02-08:
I think you are misusing the complement. The idea of auto wrap label is to
automatically wrap text from one line into multiple lines. It's not
intended for you to add line breaks in the value.

Impartcho, 2021-02-08:
I understand but I think it should, it's more a little hack than a misusing. It would natively permit to add description text directly next to the circuits and components when needed, without the need of a new text tool. "Misc/Label" does not allow that either, maybe the multilines feature would be more useful in Label, maybe in both Label and Auto-Wrap Label, maybe it needs the creation of a new tool/component.

After some researchs, it seems like I want the JTextField to be a JTextArea (one doesn't allow more than one line, the other does). Maybe you know if the component's value absolutely has to be a JTextField (if it is)? or where I can start to search to modify the components?
I found emptiness in a function in "diy-layout-creator-master\src\org\diylc\core\annotations\MultiLineText.java" but I seriously know nothing to Java so I'll need more time to understand any of your code.

Anyway thanks for the answers, if you got anything (tool, link or advice) that can help me save some time to solve this problem and maybe add new plug-ins in the future, I'd be really happy.

bancika, 2021-02-08:
I'll add multi-line support for the next version. If you appreciate the effort, feel free to donate for the cause :)

[image]

closed 2021-02-08 by bancika