bancika/diy-layout-creator #631: Adding custom Proto board

dhaillant, 2021-02-08:
Hi,

I'm the designer of the "Eurorack Stripboard PCB" (https://www.thonk.co.uk/shop/david-haillant-euro-stripboard-pcb/)

I saw that you already added the awesome Kristian Blasol's Protoboard and I wanted to know if it was possible to also create a custom board for my Eurorack Stripboard?
What would be the steps to add such board to DIYLC?

Thank you very much for your help!
David

bancika, 2021-02-08:
Hi David,

There are few ways to get a custom component created and included in DIYLC

- you can write Java code for the component yourself and submit a pull request, then I include it in the next release for free after reviewing the code and making sure it's up to standards
- you can hire someone to do the coding, the rest is the same
- you can hire me to do all that...the rates depend on component complexity

Cheers,
Bane

dhaillant, 2021-02-08:
Hello Bane,Thank you very much for your answer.I would be interested in hiring you for this...
Cheers,
David

bancika, 2021-02-08:
Ok, please get in touch via bancika@gmail.com. I will need to know which pads are connected together.

Cheers,
Bane

closed 2021-02-14 by bancika