bancika/diy-layout-creator #633: Perfboard not rendering correctly on initial load from saved file

chipshorter, 2021-02-17:
When a previously saved design on perfboard is loaded, the perfboard is not rendered correctly on the bottom row (see attached pic)

Tested V4.17.0 on Windows 10
Tested V4.16.0 on OSX

[image]

EL34xyz, 2021-03-08:
I just created and saved a file with a perf board
Then closed and opened DIYLC and loaded my file

The perf board displays just fine on W10 -4 4.17.0

[image]

EL34xyz, 2021-03-08:
Upload your file here so people can test your file

chipshorter, 2021-03-10:
It's a bit tricky to replicate, as it doesn't always happen. Unfortunately I haven't been able to determine the conditions, but the attached file seems to reproduce the error consistently. You can adjust the perfboard so it looks OK and re-save it, but it fails to render correctly the next load. Same issue on OSX V4.17.0, V4.16.0 and Windows 10 4.17.0. On other files I had success with re-saving, but it's not always the case.

[ev_example.diy.zip](https://github.com/bancika/diy-layout-creator/files/6112113/ev_example.diy.zip)

EL34xyz, 2021-03-10:
Here I have resized an eyelet to be about the size of the perf board holes
Then I pasted those eyelets on top of a few of your connection points

After you pasted the eyelets everywhere, you can delete the perf board and add a blank board

[image]

EL34xyz, 2021-03-10:
Here I added several eyelets and then deleted the perf board
Then I added a blank board

Tip: You can select every item of a type in the tree menu
So if you wanted to select all the eyelets and send them backwards, you can do that in the tree menu
Same goes for all the copper traces

The board is the farthest back, then the eyelets and then all the other stuff sits on top 

[image]

bancika, 2021-07-11:
I figured it out. Somehow you got your board to be a tiny bit over 43.18mm, probably due to rounding of numbers, and the code interpreted that as if there was enough room to squeeze in another row of pads.

[image]

Fixed the issue and will release as part of 4.18

closed 2021-07-11 by bancika