bancika/diy-layout-creator #637: MAC OS Big Sur

wsscott, 2021-03-17:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

wsscott, 2021-03-17:
I've got a new 2020 MacBook Pro Intel version,running Big Sur, and recently installed DIYLC on it.  I was working on a project yesterday, and everything was incredibly slow.  When I selected a component to add I would click it numerous times with no response, it reacted the same way to everything I was doing.  I ran the same program with the same project on my PC Windows 10 with no problems. What's the issue?

bancika, 2021-03-23:
Try disabling Hardware Acceleration from the Config menu.

wsscott, 2021-03-27:
That seems to have helped.  I'm cautiously optimistic!

bancika, 2021-03-28:
Cool. That option speeds up things on many system, but then slows down in some rare cases.

closed 2021-03-28 by bancika