bancika/diy-layout-creator #639: Suggestions

stevensloane, 2021-03-22:
1) Layout for shielded cable or wire 
2) Tabs for having multiple projects open
3) Ability to rename building blocks by right click

KENTUCKYTUKKER, 2021-03-27:
shielded cable can be made up using std symbol for flex wire and ad a i.e. a twisted pair for which you can choose your own colours. Even the screen can be added too

stevensloane, 2021-04-05:
I know how to find the twisted pair but cannot find flex wire or adding a screen

KENTUCKYTUKKER, 2021-04-05:
Just select another wire i.e. size AWG28 and add this to the point where the outher sheet and the twisted pair come together. I.e I used 22AWG coloured black for the main outher sheet, 28AWG twisted pair red/black for the signal cores and another 22AWG for the screen (drain) core coloured grey

baddriddim, 2021-07-28:
Hio, would ity be possible to add the basic 4p3t lorlin style footrprint layout to the software please?

bancika, 2021-07-28:
@baddriddim there is already an open request for rotary switches...it's not easy, that's why I haven't added any yet. There's so many different styles and variations.

Btw, going forward, please try to find appropriate issue or open a new one if none exist. It avoids clutter and makes things easier to track.

KENTUCKYTUKKER, 2021-07-28:
Hello Bane,

I designed my own rotary switches by using the "dot" symbol for the 
connection points and a "line" symbol for the contacts. This works fine 
as long as no dedicated symbols is available.

thanks,

René

------ Origineel bericht ------

stevensloane, 2021-07-28:
Why am I on this email?

bancika, 2021-07-28:
They hijacked your feature request :)

stevensloane, 2021-07-28:
Lol. I never got a reply on the original feature request

bancika, 2021-07-28:
It's acknowledged, but not simple to do and time is tight these days.

stevensloane, 2021-07-28:
Thanks! I understand!