bancika/diy-layout-creator #641: Auto-Isolation gap on ground fill

jongardneruk, 2021-03-28:
Hi,

A feature that I'd really _love_ would be if the ground fill tool could automatically leave an editable gap between parts and traces. 

It's possible in Eagle by:

1. Drawing a polygon
2. Giving it the same name as the ground connections on your schematic, usually ”GND”
3. Click the ratnest button to refresh the ratnest after adding parts or traces

Is this something that could be worked into a future release?

Thanks

Jon

bancika, 2021-03-28:
well that exact workflow is not possible in DIYLC for several reasons.

However, it could be possible to add a property to traces and pads that designates clearance/gap around them and paint that area white. That way, ground plan can be solid black and then traces and pads would be in charge of painting their own gaps

jongardneruk, 2021-03-28:
> well that exact workflow is not possible in DIYLC for several reasons.
>
> However, it could be possible to add a property to traces and pads that designates clearance/gap around them and paint that [...]

That would be perfect! At the moment I have to export as PDF then do it in Adobe Illustrator. To have some way of creating isolated traces and paths in DIYLC would save so much time.

bancika, 2021-03-29:
let's keep it open as a reminder until it gets into the pipeline and gets done.

jongardneruk, 2021-03-29:
> 
> 
> let's keep it open as a reminder until it gets into the pipeline and gets done.

Apolagies, I'm new to GitHub and am still learning how things work.