bancika/diy-layout-creator #642: VERY slow and laggy PLUS issues with images

cowasakiaudio, 2021-03-29:
I am using a rather powerful Windows 10 PC - i9 (x8@4.7GHz), 3070 GPU, 970 NVME SSD, 32Gb 4200 DDR4 Ram

The program runs but is extremely slow and laggy when trying to move objects.  Often when I select objects I have to wait 3-4 seconds between each attempt just in case it registers two or no clicks.  Moving an object requires stopping at it final location for 3-4 seconds and waiting in case it is still moving.  I don't have the same issues on my 2020 Macbook Pro which is less powerful albeit still quick.

PLUS

With images I can't select ANY percentage when trying to re-size images.  On my last piece of work the image was quite large but would only change size at 4% then 8% when 6% would have been better.

It's a great piece of software but a few issues

bancika, 2021-03-29:
Try to disable Hardware Acceleration in Config menu.

PS going forward, please try to file separate ticket for each issue. It's impossible to track progress of issues if they are randomly bundled together.

cowasakiaudio, 2021-03-29:
Thanks.  It was disabled.  Enabling it makes it worse.

bancika, 2021-03-29:
Does it work with any file or just with a specific one? If it's the latter,
please upload it so I can examine.

cowasakiaudio, 2021-03-29:
I've only created one file on the PC. I will check later.

bancika, 2022-11-29:
Is this still an issue? Cache Boost feature should help in cases like this

bancika, 2023-03-29:
closing due to inactivity

closed 2023-03-29 by bancika