bancika/diy-layout-creator #644: Configuration directory in home directory not hidden

fiechr, 2021-04-05:
I'm using version 4.17 on Manjaro (AUR package).

I don't remember exactly since when, but diylc does now create a regular directory in the home folder `/home/name/diylc` instead of `/home/name/.dicly`. Interestingly the directory `.diylc` still exists, but contains older config files and dirs. If I delete the new, visible folder, it gets recreated every time I start the application.

bancika, 2021-04-05:
Yes, DIYLC automatically migrates from old to new directory by design and leaves the old one there in case you decide to use older version of the app.

fiechr, 2021-04-07:
Is there a specific reason, why the config dir is now visible? To be honest, I'm not a fan of that. If a visible directory for the actual documents (circuit drawings) is created, it should be inside the Documents folder, or, if only for config and working files, it should be inside of $HOME/.config or a hidden directory inside the user's home dir (like it was before).

bancika, 2021-04-07:
Yes, as it also contains logs and backup files, it's not strictly config
dir. Many users struggled to find it to send the logs, so I switched to
visible.

gauthierdeplaude, 2021-12-22:
Hello,
Sorry to dig up quite an old post but I am also very annoyed with this cumbersome diylc folder in my home directory...
Would it be possible to let to the user the possibility to rename this folder from  $HOME/diylc to  $HOME/..diylc (or any other location like .config etc like suggested above)?
Thanks!

Chrismettal, 2022-05-04:
Hey, also bumping this issue. 

I don't think software on Linux should ever create visible folders under `~/` for configurations. Basically no software does that. That's what the dotfiles and dotfolders are for. 

Currently I don't see the problem with just putting the logs under `~/.diylc`. Maybe the actual problem is just documenting clearly where to find the folder in case logs are needed?

fiechr, 2022-05-04:
Maybe add a menu option to open the config directory or the log files. This would make it easier to access them, even in a hidden folder.

bancika, 2022-05-04:
It's already there under Help

Chrismettal, 2022-05-04:
That's awesome! So with this there is no reason for the config folder to not be hidden on Linux systems, right?

bancika, 2022-05-04:
Few reasons: Backward compatibility. Every time I move stuff, people lose
their configuration. Alsl, Diylc is built for all platforms, so I generally
tend to avoid doing platform-specific stuff.

Is that one directory reeeealy that big of a deal?

Chrismettal, 2022-05-04:
Yes, it is. 

The Linux Filesystem hierarchy Standard even specifies where which kind of config files should be put so that no matter what app the user installs, they will know where to expect and find what. 

https://refspecs.linuxfoundation.org/FHS_3.0/fhs-3.0.pdf

3.8.2. explains how user specific configurations under  `~/` should be handled. 

I totally understand the desire for platform-specific changes to be avoided when going cross-platform. But some conventions should still be kept. I assume under Windows DIYLC will not create a `C:\home\diylc` to store configs in "the same" path as Linux anyway, right? It will probably put it under the users `Documents` or `$Appdata$` I would guess?

(Also just anectdotal, if even half of my installed apps went "One folder isn't that big of a deal" I would have my actual home contents buried between 41 app specific directories on my main system.)

gauthierdeplaude, 2022-05-05:
Thanks @Chrismettal for that very clear explanation. 

I finally found the following workaround to get rid of that offending ~/diylc folder : 

Update run.sh in diylc root folder by moving ~/diylc to ~/.diylc on closing java app and moving ~/.diylc to ~/diylc before launching java app. For example: 

`mv ~/.diylc* ~/diylc && java -Xms512m -Xmx2048m -javaagent:lib/jar-loader.jar -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib  --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/java.text=ALL-UNNAMED --add-opens java.desktop/java.awt=ALL-UNNAMED --add-opens java.desktop/java.awt.font=ALL-UNNAMED --add-opens java.desktop/java.awt.geom=ALL-UNNAMED org.diylc.DIYLCStarter "$@" && mv ~/diylc ~/.diylc`

Not very proud of that unsatisfying solution and I hope a next release will fix this issue.

M0JXD, 2025-11-04:
> Thanks [@Chrismettal](https://github.com/Chrismettal) for that very clear explanation.
>
> I finally found the following workaround to get rid of that offending `~/diylc` folder :
>
> Update `run.sh` in [...]

Now that the AppImage is the primary distribution method for DIYLC this isn't an option anymore for a lot of users, and is agreeably unsatisfying as if a user leaves DIYLC open to use other apps or just organise files the folder persists. I would also much prefer DIYLC to hide it's config folder, or perhaps provide the option to configure it's logs location as I can understand the reluctance after users have struggled to provide logs in the past. 

As an idea of how this could be achieved maybe DIYLC could check for the existence of the hidden folder and a file/marker inside it, e.g. `USE_HIDDEN_DIR` or whatever. Thanks.