bancika/diy-layout-creator #645: Black Color Not Rendering Correctly In Some Apps

robrobinette, 2021-04-06:
.png files created by DIYLC x64 are not rendered correctly in Chrome web browser, Windows Paint, Windows Photos (all Win 10 x64). They are rendered correctly when opened with Irfanview x64. The problem is that solid blacks are rendered as white with a black outline. Grays are rendered correctly.

I have attached a DIYLC .png that doesn't render correctly. The title and power cord should be solid black but are white with a black outline instead. I first noticed this about 3 months ago and assumed they got corrupted somehow but I have confirmed the problem on two different Win 10 computers (one is x32). I re-installed the latest x64 version of DIYLC and the problem remains.

[image]

bancika, 2021-04-07:
Wow, interesting :) Thanks for reporting

bancika, 2021-04-07:
I tried to replicate but it doesn't happen on my machine, which tells me
that it has to do with that particular project. Can you please upload the
DIY file as well so I can examine it?

robrobinette, 2021-04-07:
How do I upload a .diy file?

I have this problem with all my new diy files, not just this one. As I mentioned I re-installed DIYLC and the problem persists.

bancika, 2021-04-07:
The easiest way to upload a file is to change the extension to .TXT and
just drop it to your response on github page.

robrobinette, 2021-04-07:
Here is the file:
[Deluxe_Micro_EF80_Layout.txt](https://github.com/bancika/diy-layout-creator/files/6272158/Deluxe_Micro_EF80_Layout.txt)

bancika, 2021-04-07:
Thanks. That's very interesting. When I export that same file on my machine it looks good.
Your file actually has transparency where the black should be.

Can you please upload the config file as well? You can get to it from DIYLC menu, Help -> Access User Files.

[image]

robrobinette, 2021-04-07:
Here you go:
[config.txt](https://github.com/bancika/diy-layout-creator/files/6272297/config.txt)

robrobinette, 2021-04-10:
I have to apologize, I just realized that this problem is with irfanview, the program I use to scale down the .png files for web use. The .png files put out by DIYLC are fine. Again, sorry for the false alarm.

closed 2021-04-10 by robrobinette

bancika, 2021-04-10:
Phew. No worries, no harm done. I was playing with your file and config but
couldn't replicate 😀

Cheers