bancika/diy-layout-creator #646: saving project files

KENTUCKYTUKKER, 2021-04-18:
Since the installation of version 4.17.0 I'm facing every now an then a problem in saving projects. .e. when I open an old project and modify this it will save this in the backup file. Sometimes when I save it as "save as" it overwrites the old file, while the other time it is indicating the file with the extension "temp". I have to go back and find it in the backup file and open the document again. This is very rare. I'm running the program version 4.7.0, yes an old version ,and are not facing this problem.
[log-2021-04-18 02.28.49.txt](https://github.com/bancika/diy-layout-creator/files/6331217/log-2021-04-18.02.28.49.txt)

bancika, 2022-11-30:
hi, is this still an issue. I still cannot reproduce it in any of the subsequent versions.

The attached log file is not the right one as it has no actions logged after the app is started.

The only time you can be left with a temp file next to the real one is when some other app is locking the main DIY file. 
DIYLC always saves to temp file then just renames it to DIY file, to make sure that the whole project is saved and no data is lost.

bancika, 2025-02-13:
closing due to inactivity

closed 2025-02-13 by bancika