bancika/diy-layout-creator #648: TO-92 in perf, v. 4.17.0

jagtone, 2021-04-20:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2021-04-21:
what do you mean by this?

htor, 2021-05-06:
have you tried disabling the option under Config > Snap To > Grid/Components, moving the component back in place and then enabling the option again?

it seems when you disable this option and move things slightly off-grid then sometimes components will not snap back to grid when enabled again.

jagtone, 2021-05-06:
Hi, Branislav. Thanks for getting back to me.
My question was about getting TO-92 transistor leads to match the standard
perfboard pattern, but I worked it out by adjusting the lead spacing in the
part, and then just copying it. This works fine, but if there is another
solution, I'd be happy to hear about it.
Regards,
Ben

bancika, 2021-07-10:
Yeah, standard perfboard is spaced at 0.1" and TO-92 is spaced at 0.05" (see below), so inherently they cannot align. Your workaround is fine. Also, using thin wires mimicking flexible leads coming from the outer leads is OK.

[image]

closed 2021-07-10 by bancika