bancika/diy-layout-creator #651: Mini toggle switch lugs won't rotate

gtranter, 2021-05-15:
Changing the orientation of a mini toggle switch from vertical to horizontal rotates the switch body but not the lugs.

[image]

v4.17.0

IzenEars, 2021-07-03:
How do you rotate it sideways at all?  It tells me it can't be rotated.  I don't care about the lugs.

bancika, 2021-07-03:
Please take a look the manual
https://github.com/bancika/diy-layout-creator/blob/wiki/Manual.md

IzenEars, 2021-07-03:
I found it thank you.  It was in the double click menu for the switch, not the right click menu.  I was confused.

bancika, 2021-07-04:
Yes, that is a bug, it should work from both places, although component
edit dialog is the main place to edit any property of a component.

Will fix it for the next version.

Cheers

bancika, 2021-07-10:
Fixed for 4.18

closed 2021-07-10 by bancika