bancika/diy-layout-creator #652: Features Request (import & edition wires)

G-Fil, 2021-06-02:
Hi,
Actually, I'm using successfully Inkscape as my layout creator/editor ... It give nice renders, but it is very slow as long as the drawing get complexity, and doesn't offer "electrical-frienly" facilities as DIYLC does ... 

1) So, if there were plans to include in DIYLC some module to import "svg" and/or "dxf", it would be great ! Particulary to import at a good scale a chassis already drawn with other software ... 
2) Actually, wires can only be edited via 2 points . It seems to me it's not enought to route it with more than 3 axes . Is planned a function allowing to add editing points on curves lines (wire) ?

Only these features are missing to convince me to migrate definitely to DIYLC, not only to try it lightly each time I know a new opus ... 

Regards

bancika, 2021-06-02:
Wires can have up to 7 control points, just double click to set the number
of control points

bancika, 2021-07-10:
Added 'SVG Image' component for 4.18 version, keep an eye on the update soon.

closed 2021-07-10 by bancika