bancika/diy-layout-creator #655: Crash on startup , archlinux

alandmoore, 2021-07-04:
Latest version of diylc crashes when starting on Arch Linux.

Logfile just contains the following:

```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=gasp
09:20:30,803 ERROR [main] DIYLCStarter - Major error while starting DIYLC
java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:679)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:456)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:381)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:356)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:53)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:46)
        at org.diylc.DIYLCStarter.initializeConfiguration(Unknown Source)
        at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @d737b89
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
        at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:177)
        at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
        at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:59)
        ... 9 more
```

Tried reinstalling, same issue whenever I start.

bancika, 2021-07-04:
Looks like it's something about java for that platform. Can you post the
complete log? Maybe updating java or trying some othet version or
installation would help. I'm not familiar with arch linux and its nuances.

alandmoore, 2021-07-04:
This is the complete contents of the log;
```
09:20:30,803 ERROR [main] DIYLCStarter - Major error while starting DIYLC
java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:679)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:456)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:381)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:356)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:53)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:46)
        at org.diylc.DIYLCStarter.initializeConfiguration(Unknown Source)
        at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @d737b89
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
        at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:177)
        at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
        at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:59)
        ... 9 more
```

Here's my java details from java --version:
```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=gasp
openjdk 16.0.1 2021-04-20
OpenJDK Runtime Environment (build 16.0.1+9)
OpenJDK 64-Bit Server VM (build 16.0.1+9, mixed mode)
```

This is the latest openjdk runtime for arch.

bancika, 2021-07-04:
Found something on google, can you try to add these params to java call in
run.sh (assuming that's how you run it)

--add-opens java.base/java.lang.reflect=ALL-UNNAMED
--add-opens java.base/java.util=ALL-UNNAMED
--add-opens java.base/java.text=ALL-UNNAMED
--add-opens java.desktop/java.awt.font=ALL-UNNAMED

alandmoore, 2021-07-04:
Thanks. I added those in run.sh, but it did not change the result.  I've also tried running it in openjdk 11.  No difference.

What's weird is, it was working fine for a couple months, then just recently this crash started.

bancika, 2021-07-04:
Strange. I tried updating one of the libraries that throws that error, can
you try this build and see if that helps
http://diy-fever.com/temp/diylc-4.17.0.zip

Tnx

alandmoore, 2021-07-05:
I got a lot more output from this one, though it still locks up during startup.  Logfile here:

```
23:36:53,911  INFO [main] ConfigurationManager - Initializing configuration for: diylc
23:36:54,082  INFO [main] ConfigurationManager - Saving configuration
23:36:54,134  INFO [main] DIYLCStarter - JarLoader strategy: agent
23:36:54,134  INFO [main] DIYLCStarter - Loading languages...
23:36:54,134  INFO [main] LangUtil - Loading translation for English
23:36:54,135 DEBUG [main] LangUtil - Loading language file: /home/alanm/iylc/lang/English.txt
23:36:54,142 DEBUG [main] DIYLCStarter - Java version: 16.0.1+9 by Oracle Corporation
23:36:54,143 DEBUG [main] DIYLCStarter - Java home: /usr/lib/jvm/java-16-openjdk
23:36:54,143 DEBUG [main] DIYLCStarter - OS: Linux 5.10.43-1-lts
23:36:54,143  INFO [main] DIYLCStarter - Starting DIYLC with working directory /home/alanm/iylc
23:36:54,143  INFO [main] DIYLCStarter - Configuration dump start.
23:36:54,143  INFO [main] DIYLCStarter - hiQualityRender = true
23:36:54,143  INFO [main] DIYLCStarter - metric = true
23:36:54,143  INFO [main] DIYLCStarter - snapTo = Components
23:36:54,143  INFO [main] DIYLCStarter - continuousCreation = false
23:36:54,143  INFO [main] DIYLCStarter - outline = false
23:36:54,143  INFO [main] DIYLCStarter - highlightContinuityArea = false
23:36:54,143  INFO [main] DIYLCStarter - hardwareAcceleration = true
23:36:54,144  INFO [main] DIYLCStarter - defaultTemplatesImported = true
23:36:54,144  INFO [main] DIYLCStarter - defaultBlocksImported = true
23:36:54,144  INFO [main] DIYLCStarter - Configuration dump end.
23:36:54,155  INFO [main] DIYLCStarter - Creating the main frame...
23:36:54,157  INFO [Thread-1] FontOptimizer - Starting font optimizer
23:36:54,224 ERROR [main] Presenter - Could not find version number, using default
com.thoughtworks.xstream.converters.ConversionException: No converter available
---- Debugging information ----
message             : No converter available
type                : java.util.Arrays$ArrayList
converter           : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
message[1]          : Unable to make field protected transient int java.util.AbstractList.modCount accessible: module java.base does not "opens java.util" to unnamed module @12d2ce03
version             : 1.4.17
-------------------------------
	at com.thoughtworks.xstream.core.DefaultConverterLookup.lookupConverterForType(DefaultConverterLookup.java:88)
	at com.thoughtworks.xstream.XStream$1.lookupConverterForType(XStream.java:485)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:56)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1431)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1411)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1305)
	at org.diylc.presenter.Presenter.<clinit>(Unknown Source)
	at org.diylc.swing.gui.MainFrame.<init>(Unknown Source)
	at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
23:36:54,233  INFO [main] Presenter - Checking if variants need to be updated
23:36:54,233  INFO [main] Presenter - Loading component types.
23:36:54,234 DEBUG [main] Utils - Scanning jar: /home/alanm/iylc/lib/diylc-core.jar
23:36:54,244 DEBUG [main] Utils - Scanning jar: /home/alanm/iylc/library/main.jar
23:36:54,351  INFO [main] Presenter - Loading additional library JARs
23:36:54,352 DEBUG [main] JarScanner - Found JAR file: main.jar
23:36:54,352 DEBUG [main] JarScanner - Scanning main.jar
23:36:54,381  INFO [main] ClassLoaderUtil - Adding file to the classpath: /home/alanm/iylc/library/main.jar
23:36:54,383 DEBUG [main] JarScanner - Loaded class: org.diylc.components.Abstract3LegSymbol
23:36:54,383 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractLeadedComponent
23:36:54,383 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractMultiPartComponent
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractRadialComponent
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractSchematicLeadedSymbol
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractProtoBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractVeroBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.BlankBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.Breadboard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EurorackStripboard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EyeletBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.MarshallPerfBoard
23:36:54,384 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.PerfBoard
[... 148 lines cut ...]
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
[... cut ...]
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
	at com.thoughtworks.xstream.core.util.Fields.locate(Fields.java:40)
	at com.thoughtworks.xstream.converters.collections.TreeMapConverter$Reflections.<clinit>(TreeMapConverter.java:135)
	... 19 more
23:36:55,149  INFO [Thread-1] FontOptimizer - Font optimization completed in 704ms. Found 3 slow fonts
23:36:55,149 DEBUG [Thread-1] FontOptimizer - Slow fonts: Z003,Dialog,Yeseva One
23:36:55,149  INFO [Thread-1] ConfigurationManager - Saving configuration
```

alandmoore, 2021-07-08:
Got a chance to play with this some more, downgrading to openjdk 15 seems to fix the problem.  Version 16 is where the problems begin.  Not sure what this means for other runtimes on other platforms.

bancika, 2021-07-08:
Ha, thanks for the update, good to know.

bancika, 2021-07-08:
That helped looking in the right direction. They changed something in Java
16 so now we have to use another argument to allow the code to do what it
does. It's the same in windows with java 16.

--illegal-access=permit

To make things worse, it breaks the code when used with older Java...so I
have to make two versions for runner.

Cheers

fatboab, 2022-01-30:
Just downloaded the latest zip file; turns out I didn't have Java installed, so I also installed `openjdk-17-jdk`. When I try and run `./run.sh`, I get:
```
diylc-4.20.0$ ./run.sh 
09:45:08,761 ERROR [main] DIYLCStarter - Major error while starting DIYLC
java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:679)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:456)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:381)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:356)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:53)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:46)
        at org.diylc.DIYLCStarter.initializeConfiguration(Unknown Source)
        at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @1623b78d
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
        at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:178)
        at java.base/java.lang.reflect.Field.setAccessible(Field.java:172)
        at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:59)
        ... 9 more
Major error while starting DIYLC
java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:679)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:456)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:381)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:356)
        at org.diylc.appframework.miscutils.ConfigurationManager.<init>(ConfigurationManager.java:53)
        at org.diylc.appframework.miscutils.ConfigurationManager.getInstance(ConfigurationManager.java:46)
        at org.diylc.DIYLCStarter.initializeConfiguration(Unknown Source)
        at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
        at org.diylc.DIYLCStarter.main(Unknown Source)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @1623b78d
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
        at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:178)
        at java.base/java.lang.reflect.Field.setAccessible(Field.java:172)
        at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:59)
        ... 9 more
```
I noted that there maybe another parameter required, so added it to the script, but that throws:
```
OpenJDK 64-Bit Server VM warning: Ignoring option --illegal-access=permit; support was removed in 17.0
```
Does this mean I have to downgrade to JDK15, or is there another solution...? I'm on _Kubuntu 21.04_ if that makes any difference...

bancika, 2022-11-28:
Is this still an issue? It shouldn't be as run.sh now has --add-opens directives

closed 2022-12-14 by bancika