bancika/diy-layout-creator #662: File saving severely messed up (Linux Mint)

holmesrichards, 2021-08-18:
Severe reproducible error: Project work is lost.

In ~/.diylc/logs, the most recent log file is from last January.

diylc Version: 4.17.0 (flatpak)
OS version: Linux Mint 20.2 Cinnamon

If I go to try to save into a directory under my home directory, the file save dialog shows only a few folders. There are a lot of other folders in my home directory which I can't access in this dialog.

[image]

So I save instead into my home directory. diylc accepts this and shows the file in the title bar as being in my home directory.

[image]

However there is in fact no such file in my home directory, or (based on running the shell find command) anywhere else in my files.

[image]

Despite this, if I've left diylc running, the file does appear as in my home directory in the open file dialog, and it successfully opens my project; likewise if I select it from the recent files menu. However, if I quit out of diylc and restart it, then the file does not appear in the open file dialog. It does appear in the recent files menu but I get an error if I try to open it.

In other words, my work, which diylc claimed to have saved, has been lost.

bancika, 2021-09-06:
Thanks for reporting. It has to do with flatpak distribution. It prevents the app from writing to the home directory (while allowing Documents and such). If you need that feature, I suggest switching to the universal distribution that uses run.sh to run the app until I figure it out

holmesrichards, 2023-09-12:
Two years later this severe bug is still a problem. I just lost about an hour's work because of it. Meanwhile I have other Flatpak apps that have no trouble writing to any of my directories out of the box, and when they say they have saved to the home directory they really have.

bancika, 2023-09-13:
As mentioned earlier, it's not a DIYLC issue per se, but a problem flatpak configuration. I'm not a flatpack expert, so if there's anybody who can shed light on how to give the app rights to write anyone, I'd gladly take that advice. Flatpack creates a sandbox environment for the app and limits access. I'm sorry that you lost your data, but I offered a workaround in the previous response. Universal distribution that doesn't use flatpak has no issues like this.

holmesrichards, 2023-09-13:
Given the severity of this bug, I would advocate rescinding the [installation instructions'](https://github.com/bancika/diy-layout-creator/blob/wiki/Manual.md#installing-the-app) recommendation of the FlatPak version for Linux users. Indeed I would advocate specifically warning against its use.

bancika, 2025-03-03:
Version 4.40 I just published should be able to have access to home directory as well.