bancika/diy-layout-creator #666: Suggestion: Adding layer selection control to wiring entities for selection between Component and Board layer

StratocasterPlayer, 2021-08-24:
Hello,

First of all thank you for developing and supporting the DIY layout creator tool.

When using the tool for designing a board layout I am missing an easy way to view the board including the interconnection wiring between the eyelets on the board. This specific view shows exactly what has to be done on the board prior to adding the circuit components and other wiring.

The only way, I am aware of, to obtain this view is to copy the original diy file and manually delete the circuit components and other wiring connected to the board.
I suggest to add an option to add layer control to the relevant wiring entities in order to select between Board and Component layer. In this way everything can be easily handled in just one diy file.

Please let me know if there is already an easy way in the tool which provides the required view.

Thanks for your attention.

bancika, 2021-09-06:
There is already support for this, just go to "Layers" menu and hide layers you don't want to see.

StratocasterPlayer, 2021-09-07:
Hi Bane,

Thanks for your response.

I have noticed if the wiring between eyelets are made with hookup wire then
it's not visible when the Components layer is set to hide.
However if the "wiring" is made of a curved trace then it can be made
visible with the board only, due to a separate Trace layer. This works for
me!

Thanks and regards,

John

bancika, 2021-09-19:
Hmm...that shouldn't work like that. I'll fix it for the upcoming version...will merge traces and wires into a single layer called "Wiring" and separate wires from components.
Cheers

closed 2021-09-19 by bancika