bancika/diy-layout-creator #667: Saving Error in portable version

Noah-West, 2021-09-02:
When running the portable on windows 10, I am not able to save, and it shows the error "Could not save the project to file. Check log for details." 
This error doesn't show up when running the windows installer version. I cannot find a log file in the portable version; it does not save to the log folder that the installed version uses, and I cannot find any logs in the programs home folder.

bancika, 2021-09-06:
which portable version are you referring to?

Noah-West, 2021-09-06:
I downloaded the latest portable release, 4.19.0, from the releases page.
[diylc-4.19.0.zip](https://github.com/bancika/diy-layout-creator/releases/download/v4.19.0/diylc-4.19.0.zip)

bancika, 2021-09-06:
Oh I get it. This is actually a windows 10 security feature to block access
to most directories for apps that are executed outside of Program Files.
You can save to My Documents and similar, but not elsewhere.
One way to get around it is to run DIYLC in Administrator Mode.

bancika, 2021-09-06:
The other is to try moving it to Program Files

Noah-West, 2021-09-06:
Ok, thanks. I was wondering if it might be something like that

closed 2021-09-06 by Noah-West