bancika/diy-layout-creator #670: Suggestion - 1/8" Cliff jacks to match David Haillant Eurorack DIY PCB

Andy2No, 2021-09-20:
I found this application via the David Haillant website.  I'm not sure how long ago his DIY Eurorack PCB was added, but I don't see any suitable jacks in the menu.

The recommended jacks are Cliff CR1283 or Cliff FCR1281.  I've only managed to buy the FCR1281 sort, so far.  They're shown on the Cliff website, here, under Cliff S6 series:

https://www.cliffuk.co.uk/products/jacksockets/js2535.htm

bancika, 2021-09-21:
Hi Andy. As you noticed, there are no such components available yet in DIYLC. The engagement with David Haillant only included the said PCB, nothing else. If you with to expedite the process, you can sponsor the component and have it implemented sooner. Otherwise, things get picked up from the queue when I have free time and depending on how many people request/vote for a new feature.

Andy2No, 2021-09-21:
Thanks.  I'll bear that in mind.  I've only briefly tried DIYLC, so far, and haven't quite got it to work properly, yet, so it remains to be seen if I can get much use out of it - I have a problem with context menus persisting instead of disappearing after use.  I'll start a new issue if I can't figure it out.

bancika, 2022-12-15:
Added for 4.35

closed 2022-12-15 by bancika