bancika/diy-layout-creator #671: Cannot get backgroud to grey editing theme

visuttaja, 2021-09-20:
I need grey background for my white allergy eyes. Current  dark theme covers black font from files like this:
https://robrobinette.com/images/Guitar/5E3P_Build/5e3_Tube_Guitar_Amplifier.diy
Blue theme also messes up with black grid and hurt eyes immediately. 
I have been working on this quite a while on web pages and have found simpole solution to use just a little light grey  background which lets original black text cut through the image. Nothing else is needed.
Just simply little shade of grey as background. Color is exactly  r=170 g=170 b=170. Point is that if someone has used black font color it still can be seen very well on such grey backgourd. 
So I tried to edit those themes but it is not happening. Installation directory has config.xml which also has section for colors. 
Themes directory contains *.xml for three themes.
User directory C:\Users\<user>\.diylc has one config xml which has colors section. 
But it seems that there is also colors hardcoded in some java source because changed made in any of those given xml sections have no effect none. So there is at least four places....
I could compile the program from source but I would not like to put all my energy on it only because of need to get grey background.  Could you tell how it is choosing between those different places so I could get my grey background. I really would like to start use diylc but I just currently cannot.

bancika, 2021-09-20:
What about other colors? Grid lines? Would this work?

[image]

visuttaja, 2021-09-20:
Yes .That looks ok. My experience is that with grey starting point is possible to get most color combinations. It is overwhelmingly most unlimiting color to build on functional color schemes. If the grey is quite light at the point where it is just starting to turn little dark it takes a lot of work away and lets ideas evolve many times so that color appears not to be limiting factor. I have spend hundreds of hours to fiddle with these and now I always start with grey. An example of an additional arbitrary advantage of such grey background is that it does not suck all the ink from printer which black background would do in accidental situation.

bancika, 2021-09-21:
[image]

Played with the colors a bit and liking this combo

visuttaja, 2021-09-21:
That is very good grid.  Rare gem i would say.  I could stare that combination 16 hours which is how things usually turn out when interface allows it.

bancika, 2021-12-19:
This will be released in 4.21

closed 2021-12-19 by bancika