bancika/diy-layout-creator #675: Cutting veroboard trace

visuttaja, 2021-11-02:
I just wonder if there is any trick to cut vero board traces when I insert DIP8 opamp i.e IC. When I press "show connected areas" it goes right under the IC to other side on vero board. What I  needed in this situation was the ability to cut vero the trace under the IC  and start from there to add other components. Vertical is the natural orientation of traces when working with vero boards and ICs like njm4556,opa2134.
That is because we want to DIP8 pins to be available as continuous traces.
[image]

visuttaja, 2021-11-02:
I find that there is such a thing (connectivity/trace cut)but unfortunately it does not work. It "forgets" the cuts quite immediately. Maybe there is some trickery to get it working? Is the cut supposed to be over holes or  between holes?
Does the size matter?  Does it connect networks or through resistors too?
Here is cuts in both sides of IC under resistors. For some reason results change with every click.
In this is example cuts under resistors should count.
[image]

visuttaja, 2021-11-02:
Solution? I found that maybe it works if I set the cut circle on the hole. So not check "cut between the holes" gives:
[image]

bancika, 2021-11-03:
Hi Guster, can you please upload a DIY file that shows a problem? You may
need to change the extension to TXT if github rejects it.

Thanks

visuttaja, 2021-11-03:
Thanks Bane for you kind answer. For now I think I must learn little more bit using DIYLC so I can deliver you better quality issue communication.

bancika, 2022-11-29:
Hi, is this still an issue?

mark-cooke, 2022-12-22:
Hi @bancika - I'm (a new user) using the latest release (4.34.0) and it does seem to still be a problem.

A simple way to reproduce it is to:
1. Create a new project with the default veroboard
2. Add a vertical line of trace cuts to effectively split the board into 2 halves
3. Observe apparent connections right along the broken trace:
[image]
It seems there's a very tiny strand of persistent bit of copper somewhere around the cut ;-)

Changing the "cut between holes" option makes no difference.
I also tried increasing the size of the cut to 0.12" but that didn't help either.

bancika, 2022-12-23:
Hi, this seems to be a new bug introduced one or two versions ago. Fixed it for 4.35.

Closing

closed 2022-12-23 by bancika

mark-cooke, 2022-12-23:
Maybe I'm misunderstanding how it should work but I've just cloned the repo (as I was eager to get my personal project progressed) and built and run the latest code and it still seems to highlight unconnected areas for me.

In the image below I clicked on row B of the veroboard - my understanding is that the portion of strip from A1 to A6 shouldn't be shown as connected.
[image]

I'm fairly sure I'm running the latest version and not loading old code, as evidenced by the about screen in the shot and the fact that I see a bunch of new components in the tree on the left of the screen.

bancika, 2022-12-23:
I didn't originally push the commit. Now I did..

mark-cooke, 2022-12-23:
I've done that a few times myself....

It's perfect now - Thank You!

Congratulations on a great piece of software!