bancika/diy-layout-creator #679: New component Voltage regulators

barremans, 2021-11-24:
It would be great to have voltage regulators with 3 Pins max width 3 holes.
Changeable  pin name  because example
LM7805 has pins 
1 = VIN
2 = GND
3 = VOUT

the LD33CV has
1 = GND
2 = VOUT
3 = VIN

I could use a transtistor but its not possible
* to scale to 3 holes
* change pin names

Nice project/application by the way ;)
love it

overhumanalien, 2021-11-30:
Double click a transistor to open the properties dialog. Set the "Pin Spacing" parameter to 0.1" to distribute the legs over three rows. In the same dialog there is also "Pinout" dropdown. Although there is no I/O/G option there.

bancika, 2022-02-17:
Adding the following - I/G/O (LM780x), G/O/I (LD33CV) and O/G/I (LM780xL)...any other regulator pinouts I should add?

closed 2022-02-21 by bancika