bancika/diy-layout-creator #682: Zooming on a MacBook Pro with trackpad slow and unpredictable

overhumanalien, 2021-12-10:
Zooming with trackpad on MacBook Pro is slow and unpredictable. 

I have attached two screen capture videos where one is showing the current zoom behaviour on DiyLc and the other is the expected behaviour as seen in Illustrator. 

- Illustrator has a smooth zoom while DiyLc is jumping through the predefined zoom levels. 
- Illustrator centres the zoom to the cursor while DiyLc's zoom is following somewhat (seemingly) odd path.

https://user-images.githubusercontent.com/95300803/145588782-21f67e0c-ee4b-4adf-9a8d-72175fcb1c2c.mov

https://user-images.githubusercontent.com/95300803/145588807-16557b0f-3cae-43ed-9f66-2fbc3bee3b48.mov

babajoux, 2021-12-12:
I have the same problem. Its really sluggish and slow and freezes.

frnktrpz, 2022-01-30:
Have you tried changing the anti-aliasing, high quality rendering and hardware acceleration settings?

overhumanalien, 2022-01-31:
> Have you tried changing the anti-aliasing, high quality rendering and hardware acceleration settings?

Yes, I have tried them. None of them made much difference to performance.

What I have find out though is that as long as you have a board or a component selected zoom has predictable behaviour. The above description only appears if nothing is selected. So in the end this is not a show stopper I thought it was. More of a usability glitch that can be lived with as long as you remember to have something selected before zooming.

bancika, 2022-02-27:
I tried to improve gesture zoom on mac, but it's not any better :( . Please bear in mind that comparison with Adobe products is not very fair. They are a multi-billion dollar company and I'm a guy doing this for free in my spare time. I am striving to make DIYLC as good as it can, but we all need to manage our expectations :)

Either way, thanks for the feedback, I'll keep this ticket open, maybe I have better luck in the future...

overhumanalien, 2022-02-27:
Thank you for your effort! Honestly to put things in place there are far more bugs and usability issues in Adobes’ products that I can ever find in DIYLC. Again your work is truly appreciated and to me the zoom works fine as I have learned to have something selected before zooming.

bancika, 2022-02-27:
Thank you for the kind words :)

bancika, 2022-02-27:
Please give version 4.21.4 a go when you have a chance...I made some
improvements, it does look better to me.

bancika, 2022-03-07:
Tested on a mac, looks much better to me, I think this is as good as I can get it :)

closed 2022-03-07 by bancika

babajoux, 2022-03-10:
Thank you a lot for your work and love you put into this!

overhumanalien, 2022-03-10:
Works like a charm! Thank you very much. Please have a few more cups of coffee on me.

bancika, 2022-03-10:
Glad to hear it! And thanks!