bancika/diy-layout-creator #684: diylc-4.20.0-setup.jar throws com.izforge.izpack.api.exception.IzPackException

lhanson, 2022-01-13:
I'm assuming that the "setup" JAR will run some kind of installer, but it doesn't seem to work:

```
$ java -jar diylc-4.20.0-setup.jar 
Command line arguments: 
Using laf org.pushingpixels.substance.api.skin.SubstanceMistSilverLookAndFeel
SEVERE: com.izforge.izpack.api.exception.IzPackException: java.lang.reflect.InvocationTargetException
$ java -version
openjdk version "18-ea" 2022-03-15
OpenJDK Runtime Environment (build 18-ea+15-Ubuntu-4)
OpenJDK 64-Bit Server VM (build 18-ea+15-Ubuntu-4, mixed mode, sharing)
```

I've also tried installing the Flatpak package on my distro (Pop!_OS), but found that lacking since it didn't have access to the existing DIYLC projects on my filesystem. I did get this working at least once in the past, but the installation procedure seems like it could use some polish and documentation. Happy to help on that.

bancika, 2022-01-20:
> yep, same for me

which OS are you using?

dlabaja, 2022-01-20:
Ubuntu 20.04, java 17

dlabaja, 2022-01-20:
Well, found an flatpak alternative of the exact same version and it's working

bancika, 2022-01-20:
Interesting :)

bancika, 2022-01-21:
> I'm on Pop!_OS 21.10.

It probably has to do with java or flatpack for that exotic distribution. I
suggest trying the universal package diylc-xxx.zip that can be unpacked and
run using run.sh on any unix-like machine.

Cheers,
Bane

dirkbaechle, 2022-02-16:
I also had no success in trying to get the latest v4.20.0 (unpacked from the diylc-xxx.zip, as suggested) working under my Ubuntu 20.04 LTS, while using openjdk17. As a workaround, I have now patched my "run.sh" as follows

``` bash
#!/bin/sh
export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64
$JAVA_HOME/bin/java -Xms512m -Xmx2048m -javaagent:lib/jar-loader.jar -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib org.diylc.DIYLCStarter
```

such that the openjdk11 is used instead (assuming it is installed in the system, of course). Works like a charm for me!

@bancika : I had also started to convert your whole dev stack (app-framework, http-proxy, swing-framework, jar-loader and diy components) to JDK17, using Maven for building. This includes bumping all the dependencies (like xstream) to their latest available versions and switching to log4j2, for example. This doesn't help running the v4.20.0 on my system so I won't follow this route any further, but if you should be interested in this development simply let me know. Maybe you can use the patches...

bancika, 2022-02-16:
Please try version 4.21.0 that I just published. It works on my machine
with newer java.

dirkbaechle, 2022-02-17:
@bancika I can confirm that the new v4.21.0 works fine with openjdk17 on my system when simply unzipping and starting "run.sh" (haven't tried the setup.jar). Good job!

bancika, 2022-02-17:
Glad to hear it!

lhanson, 2022-02-17:
Same result with `diylc-4.21.0-setup.jar`:

```
$ java -jar diylc-4.21.0-setup.jar 
Command line arguments: 
Using laf org.pushingpixels.substance.api.skin.SubstanceMistSilverLookAndFeel
SEVERE: com.izforge.izpack.api.exception.IzPackException: java.lang.reflect.InvocationTargetException
$ java -version
openjdk version "18-ea" 2022-03-15
OpenJDK Runtime Environment (build 18-ea+15-Ubuntu-4)
OpenJDK 64-Bit Server VM (build 18-ea+15-Ubuntu-4, mixed mode, sharing)
```

bancika, 2022-02-17:
I see, so it's probably izpack to blame. Unfortunately, there's no newer
version that that, they stopped releasing a couple of years ago.

closed 2022-11-28 by bancika