bancika/diy-layout-creator #689: After updating to 4.21.0 or 4.21.1 settings reset, custom components/favorites gone (Win10)

DoomchildXXL, 2022-02-20:
OS: Win 10 (21H2)
[log-2022-02-20 04.48.57.txt](https://github.com/bancika/diy-layout-creator/files/8104466/log-2022-02-20.04.48.57.txt)
[log-2022-02-20 04.50.26.txt](https://github.com/bancika/diy-layout-creator/files/8104470/log-2022-02-20.04.50.26.txt)

The attached logs are from immediately after installation of 4.21.0 and 4.21.1 respectively. I installed them one after another. (Had downloaded 4.21.0 on Friday, not realizing there had already been another release in the mean time.)

After installation I found the settings had reset, and faves and components I had added over time were gone from the ui. I specified "the ui", because at least the favorite (and recently used) items (as well as recent files and my custom parts) are still listed correctly in the config.xml.

bancika, 2022-02-20:
Hi, can you try to find the previous config file? It should be in diylc
folder, the most recent file with ~ in the name. It will probably be the
largest file as it contains the most information. If it's still there,
please send it over so I can test why it failed to load with the latest.

DoomchildXXL, 2022-02-20:
In C:\Users\[me]\diylc\ I've only got one config.xml without any ~. And luckily that's the one which still contains my stuff. 

There's also a way older .diylc folder, but that hasn't got anything of note in it. Only older stuff.

...Oh, wait. Did you switch back and forth between storing in folders "diylc" and ".diylc"? I put my current config file in that folder, and now it loads fine. \[T]/

bancika, 2022-02-20:
That's done few releases ago...people complained that their diylc folder is
invisible. Glad you sorted it out!

DoomchildXXL, 2022-02-20:
I guess, I should read more change logs. Damn. :)

closed 2022-02-20 by DoomchildXXL