bancika/diy-layout-creator #692: Cannot open files

thef0nz, 2022-02-25:
Tried the configuration fix but cannot open some files. Backups do not work either. 
[log-2022-02-25 01.45.31.txt](https://github.com/bancika/diy-layout-creator/files/8139122/log-2022-02-25.01.45.31.txt)

bancika, 2022-02-25:
Please attach one of the problematic files.

thef0nz, 2022-03-02:
Downloaded older version (4.16) to open it up, saved it, and then tried opening with the latest version, works fine now! *phew* 

Can still post the file tho if you want.

bancika, 2022-03-02:
Yes please. How did you create the problematic file? With new version?

thef0nz, 2022-03-02:
Found a bunch more that need fixing so I'll post one of those. They were created with an old version. The one I fixed was the latest made in 2019. The others are earlier. I remember I was able to open them up to version 4.20. 

One of these files just magically fixed itself for 4.21.4 while uploading, schematic 2, the other still gives me an error message. 

[ProTremoVerb Schematic.diy.zip](https://github.com/bancika/diy-layout-creator/files/8170719/ProTremoVerb.Schematic.diy.zip)

[BrownfaceReverb Schematic 2.diy.zip](https://github.com/bancika/diy-layout-creator/files/8170684/BrownfaceReverb.Schematic.2.diy.zip)

bancika, 2022-03-02:
Thanks for the files. I was able to figure out the root cause...there was a
problem with the toggle switch in the past and it was fixed a year ago but
it caused the issue with backwards compatibility with the old version.
I'll issue an update, you don't need to touch those files...hopefully they
will work

thef0nz, 2022-03-03:
Just tested those files and others with 4.22, everything works! Thanks!

closed 2022-03-03 by thef0nz

Alexcypher, 2022-03-04:
Hi! Same problem with 4.22
Log:
[log-2022-03-04 03.13.54.txt](https://github.com/bancika/diy-layout-creator/files/8183759/log-2022-03-04.03.13.54.txt)
Problematic file:
[dist_layout.zip](https://github.com/bancika/diy-layout-creator/files/8183742/dist_layout.zip)
Problematic file was created with 4.20

P.S. Thank you very much for DIYLC!

bancika, 2022-03-04:
Hi Alex,

Thanks for reporting. It turns out that it's a different issue but shows
itself in the same way. I'll include the fix in the next patch version.

Cheers,
Bane

Alexcypher, 2022-03-04:
Thanks a lot!