bancika/diy-layout-creator #695: DIYLC freezes

Highway3131, 2022-03-06:
Hi,
my DIYLC freezes after a certain time off inactivity or when i change the window size and...
Working with Windows 10, latest release and latest Java
Regards,
Jim - highway31

Highway3131, 2022-03-06:
Forgot... W10 - 64 bit

bancika, 2022-03-06:
Can you please attach a log from the time the freeze happened?

domdec314, 2022-07-17:
I'm having the same problem. DIYLC is freezing after being inactive for a few minutes. I've attached the last logfile. I just started the program, loaded a project, and let it sit until it froze.
[log-2022-07-17 07.11.17.txt](https://github.com/bancika/diy-layout-creator/files/9128783/log-2022-07-17.07.11.17.txt)

SawBoxJohn, 2022-07-18:
Mine has been fixed!!!  I erased the config file and it's been great ever since ..

bancika, 2022-07-18:
Did you save the config by any chance so I can examine it? I suspect it has
to do with the Cache Boost feature. Do you have it selected now? Which
version are you using?

SawBoxJohn, 2022-07-18:
I believe I did backup the file before I deleted it, I will take a look and upload a copy of both good and bad files.  This may take me a day or three.
John

bancika, 2022-07-18:
@domdec314 thank you. Based on your config, it seems that "Hardware Acceleration" is my main suspect. It was ON in the config file and default value is OFF. There might be something with that particular graphics card and Java that doesn't sit right.

SawBoxJohn, 2022-07-18:
Confirmed that in my "bad" config, the hardware acceleration is on (true). In my working config it is off...  How do I send the file?

bancika, 2022-07-18:
Through github or to my mail, bancika at gmail

SawBoxJohn, 2022-07-18:
Files emailed...
Message ID: ***@***.***>

domdec314, 2022-07-19:
> 

Thank you for helping me figure this out. I can confirm that re-enabling "Hardware Acceleration" causes it to freeze again. For reference, I'm using a GeForce GTX 970 with driver version 516.40. I'm also using Java Version 8 Update 333.

SawBoxJohn, 2022-07-19:
Fantastic news!

bancika, 2022-07-31:
I'm killing hardware acceleration feature as it causes several issues with modern cards/drivers.

closed 2022-07-31 by bancika