bancika/diy-layout-creator #696: Issues with 4.22.1 not being able to import building blocks.

EL34xyz, 2022-03-11:
This was reported on my forum and on facebook by mac and windows users

I tested importing variants and I was able to import them on Windows 10 64 bit using DIYLC 4.22.1

I tested building blocks and this operation fails

Looking at my log file, it looks like ellipse's are causing problems when importing building blocks?

Excerpt from the log file shown below

```

10:34:40,184 DEBUG [SwingWorker-pool-1-thread-4] Presenter - importBlocks(C:\Users\hoffm\Desktop\HoffmanBuildingBlocks.xml)
10:34:40,268 ERROR [AWT-EventQueue-0] MainFrame - Background task execution failed
java.util.concurrent.ExecutionException: com.thoughtworks.xstream.converters.ConversionException:
---- Debugging information ----
cause-exception : com.thoughtworks.xstream.mapper.CannotResolveClassException
cause-message : diylc.shapes.Ellipse
class : java.util.HashSet
required-type : java.util.HashSet
converter-type : com.thoughtworks.xstream.converters.collections.CollectionConverter
path : /org.diylc.common.BuildingBlockPackage/block/entry/set/diylc.shapes.Ellipse
class[1] : java.util.HashMap
required-type[1] : java.util.HashMap
converter-type[1] : com.thoughtworks.xstream.converters.collections.MapConverter
class[2] : org.diylc.common.BuildingBlockPackage
required-type[2] : org.diylc.common.BuildingBlockPackage
converter-type[2] : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
version : 1.4.19
```

EL34xyz, 2022-03-11:
The full log file is attached

[log-2022-03-09 10.34.28.txt](https://github.com/bancika/diy-layout-creator/files/8233540/log-2022-03-09.10.34.28.txt)

bancika, 2022-03-11:
can you attach building blocks file please?

EL34xyz, 2022-03-11:
Yes, here is mine that I tested with a clean install of DIYLC and a new config
I don't know what the others are trying to import?

[HoffmanBuildingBlocks.zip](https://github.com/bancika/diy-layout-creator/files/8233898/HoffmanBuildingBlocks.zip)

bancika, 2022-03-12:
Seems like a part of it is a change made almost two years ago and part is a recent change. I had to update your file to replace all "set" tags with "list", but I have to release a patch before it can work...here's the updated file

[HoffmanBuildingBlocks.zip](https://github.com/bancika/diy-layout-creator/files/8238348/HoffmanBuildingBlocks.zip)

closed 2022-03-12 by bancika

EL34xyz, 2022-03-12:
Thanks
I do remember the multi cap can deal a while back
My file was later than that fix

I can wait for the patch and then export all my variants and building blocks for other to use.