bancika/diy-layout-creator #698: diylc-4.22.2 Java Virtual Machine Launcher error

blackcorvo, 2022-03-29:
I just downloaded the latest .zip version of the software, and trying to run the .exe file results in the following error:

[image]

It still opens if you run the .jar file, so it seems like an issue specifically with the executable. And the older 4.20.0 .zip I have here runs completely fine, so it's not an incompatibility with my system.

blackcorvo, 2022-03-31:
After messing with the files a bit, I noticed that copying the 4.20.0 .ini file to the new version's folder made the program run normally.
Opening both .ini files on notepad revealed that removing the highlighted code from the 4.22.2 .ini allows the application to initiate properly. This code is not present in the 4.20.0 .ini file.
[image]
Not sure if it's some test code that was left over by accident, I don't know Java, but I'll be running it with the code removed for now.

bancika, 2022-03-31:
Thanks for reporting!

bancika, 2022-04-23:
Fixed

closed 2022-04-23 by bancika